Seeed-Studio/platform-seeedboards #1: Seeed XIAO nrF52840 Sense break Serial (not recognized)

JpEncausse, 2025-04-16:
Hello,
I can't get serial data on PlatformIO with Seeed XIAO nrF52840 Sense. The problems seems to comes from the way PlatformIO upload the firmware. It works on ArduinoIDE

Here is my PlatformIO configuration : 

```
[env:seeed_xiao_nrf52840_sense]
platform = https://github.com/Seeed-Studio/platform-seeedboards.git
board = seeed-xiao-afruitnrf52-nrf52840-sense
framework = arduino
build_flags = 
	-w
	-DBOARD_HAS_PSRAM
	-DARDUINO_USB_CDC_ON_BOOT=1 ; Using Zadig to reinstall Driver
	-DCORE_DEBUG_LEVEL=1
```

And a very simple code : 

```
#include <Arduino.h>
#include <Adafruit_TinyUSB.h>

void setup() {
  Serial.begin(115200);
  while (!Serial);  // Attendre la connexion série
  Serial.println("USB CDC actif");
}

void loop() {
  Serial.println("Ping");
  delay(1000);
}
```

✅ Using latest version of Arduino IDE it works and serial communication display th Ping. Using Zadig I see a TinyUSB device. Perfect !

❌Using PlatformIO the code is correctly pushed, installed, and run BUT the serial communication is broken, VSCode do not see board/port and Zadig see 2 nRF device (Interface 0 and 2)

I assume PlatformIO is doing something Wrong but ChatGPT loop with wrong advices ... The board and setup seems ok. Any ideas ?

ackPeng, 2025-04-23:
The content in the post seems to show that the problem has been solved. Do you have any other questions?

ThePradox, 2025-05-09:
I can confirm that this is still a problem! And a very confusing one, especially if you come from the wiki (https://wiki.seeedstudio.com/xiao_nrf52840_with_platform_io/).
Without `lib_archive = no` in `platformio.ini` serial will not work! The nrf52840 will not be visible as a COM port, it's not possible to upload a sketch without double-pressing in bootloader mode, and `while (!Serial);` will never complete.

While `lib_archive = no` works, the previous board definition from the wiki (https://github.com/maxgerhardt/platform-nordicnrf52) also works fine.

ackPeng, 2025-05-14:
This seems to be a compatibility issue with the library. We will study some solutions to solve it.

JpEncausse, 2025-05-14:
Very curious if ther eis some workaround to connect to Bluefruit
https://stackoverflow.com/questions/79584991/bluefruit-no-arduino-access-to-service-or-characteristic-of-nrf52840-devices

ackPeng, 2025-06-04:
[image]

closed 2025-06-26 by github-actions[bot]