Seeed-Studio/platform-seeedboards #2: Seed XIAO ESP32 C6 fails to build on macOS

aparcar, 2025-04-17:
I'm seeing the following compilation error and don't know how to proceed:

```
/Users/user/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: /var/folders/bb/1dd48msn3b5_nx5zw1zr3n5w0000gn/T//ccXswafB.ltrans0.ltrans.o: in function `.L0 ':
xxx/.pio/libdeps/seeed_xiao_esp32_c6/ArduinoBLE/src/utility/L2CAPSignaling.cpp:220:(.text+0x4bb0): undefined reference to `esp_vhci_host_register_callback'
/Users/user/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: /var/folders/bb/1dd48msn3b5_nx5zw1zr3n5w0000gn/T//ccXswafB.ltrans0.ltrans.o: in function `.LVL1882':
xxx/.pio/libdeps/seeed_xiao_esp32_c6/ArduinoBLE/src/utility/btct.cpp:154:(.text+0x4bce): undefined reference to `esp_vhci_host_check_send_available'
/Users/user/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: /var/folders/bb/1dd48msn3b5_nx5zw1zr3n5w0000gn/T//ccXswafB.ltrans0.ltrans.o xxx/.pio/libdeps/seeed_xiao_esp32_c6/ArduinoBLE/src/utility/btct.cpp:155:(.text+0x4be0): undefined reference to `esp_vhci_host_send_packet'
collect2: error: ld returned 1 exit status
*** [.pio/build/seeed_xiao_esp32_c6/firmware.elf] Error 1
```

aparcar, 2025-04-17:
The `platformio.ini` looks like this:

```ini
[env:seeed_xiao_esp32_c6]
platform = Seeed Studio
board = seeed-xiao-esp32-c6
framework = arduino
monitor_speed = 115200
```

ackPeng, 2025-04-18:
[image]  Hello, we have also tested it on Mac and your problem does not occur. @aparcar

ackPeng, 2025-05-07:
@aparcar  Does this use a demo or library of esp32-c6-devkitm-1?

closed 2025-05-23 by ackPeng