Seeed-Studio/platform-seeedboards #3: unable to create project in vscode > platform.io > xiao esp32-c6 with esp-idf option

robertvandervoort, 2025-04-20:
PIO Core Call Error: "in get boards\r\nThe following files/directories have been created in e:\\Users\\bigro\\Documents\\PlatformIO\\Projects\\test-esp32c6\r\ninclude - Put project header files here\r\nlib - Put project specific (private) libraries here\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\nResolving seeed-xiao-esp32-c6 dependencies...\r\nArchitecture =: esp\r\nesp configure_default_packages\r\nin get boards\r\nTrue\r\nTrue\r\nTrue\r\npop toolchain-xtensa-esp-elf\r\nAlready up-to-date.\r\nArchitecture =: esp\r\nesp configure_default_packages\r\nin get boards\r\nTrue\r\nTrue\r\nTrue\r\npop toolchain-xtensa-esp-elf\r\nUpdating metadata for the vscode IDE...\r\n\n\nUserSideException: Processing seeed-xiao-esp32-c6 (platform: Seeed Studio; board: seeed-xiao-esp32-c6; framework: espidf)\r\n--------------------------------------------------------------------------------\r\nArchitecture =: esp\r\nesp configure_default_packages\r\nin get boards\r\nTrue\r\nTrue\r\nTrue\r\npop toolchain-xtensa-esp-elf\r\nArchitecture =: esp\r\nesp configure_default_packages\r\nin get boards\r\nTrue\r\nTrue\r\nTrue\r\npop toolchain-xtensa-esp-elf\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nArchitecture =: esp\r\nesp configure_default_packages\r\nin get boards\r\nTrue\r\nTrue\r\nTrue\r\npop toolchain-xtensa-esp-elf\r\nin get boards\r\nin get boards\r\nboard id is seeed-xiao-esp32,will call board_build/esp/esp_build.py\r\nfranwork_dir = C:\\Users\\bigro\\.platformio\\packages\\framework-arduinoespressif32\r\nin get boards\r\nin get boards\r\nCONFIGURATION: https://docs.platformio.org/page/boards/Seeed Studio/seeed-xiao-esp32-c6.html\r\nPLATFORM: Seeed Studio Xiao Series (1.0.0+sha.c5cb08e) > Seeed Studio XIAO ESP32C6\r\nHARDWARE: ESP32C6 160MHz, 512KB RAM, 4MB Flash\r\nDEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)\r\nPACKAGES: \r\n - esp32-arduino-libs @ 5.1.0+sha.d38afc77db \r\n - framework-arduinoespressif32 @ 3.1.1+sha.108a0792 \r\n - framework-espidf @ 3.50302.0 (5.3.2) \r\n - tool-esptoolpy @ 4.8.7 \r\n - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 \r\n - tool-sreccat @ 1.164.0 (1.64) \r\n - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 \r\n - toolchain-riscv32-esp @ 13.2.0+20240530\r\nin get boards\r\nin get boards\r\nin get boards\r\nin _embed_files py\r\nin get boards\r\nin get boards\r\n--- preprocess_linker_file Debugging ---\r\nsrc_ld_script: C:\\Users\\bigro\\.platformio\\packages\\framework-espidf\\components\\esp_system\\ld\\esp32c6\\memory.ld.in\r\ntarget_ld_script: E:\\Users\\bigro\\Documents\\PlatformIO\\Projects\\test-esp32c6\\.pio\\build\\seeed-xiao-esp32-c6\\esp-idf\\esp_system\\ld\\memory.ld.in\r\n--- Installed Packages ---\r\n - esp32-arduino-libs\r\n - framework-arduinoespressif32\r\n - framework-espidf\r\n - tool-esptoolpy\r\n - tool-riscv32-esp-elf-gdb\r\n - tool-scons\r\n - tool-sreccat\r\n - tool-xtensa-esp-elf-gdb\r\n - toolchain-riscv32-esp\r\nDebug: Checking for cmake.exe at: bin\\cmake.exe\r\nDebug: Path exists: False\r\nDebug: Checking for cmake.exe at: C:\\Users\\bigro\\.platformio\\packages\\tool-cmake\\bin\\cmake.exe\r\nDebug: Path exists: True\r\nDebug: Is file: True\r\nDebug: Found cmake.exe at: C:\\Users\\bigro\\.platformio\\packages\\tool-cmake\\bin\\cmake.exe\r\ncmake_path: C:\\Users\\bigro\\.platformio\\packages\\tool-cmake\\bin\\cmake.exe\r\nDebug: About to call os.path.join()\r\nDebug: cmake_path: C:\\Users\\bigro\\.platformio\\packages\\tool-cmake\\bin\\cmake.exe\r\nDebug: 'bin': 'bin'\r\nDebug: 'cmake.exe': 'cmake.exe'\r\nDebug: cmake_bin_path: C:\\Users\\bigro\\.platformio\\packages\\tool-cmake\\bin\\cmake.exe\r\ncc_path: C:\\Users\\bigro\\.platformio\\packages\\toolchain-riscv32-esp\\bin\\riscv32-esp-elf-gcc\r\nconfig_dir: E:\\Users\\bigro\\Documents\\PlatformIO\\Projects\\test-esp32c6\\.pio\\build\\seeed-xiao-esp32-c6\\config\r\nld_dir: C:\\Users\\bigro\\.platformio\\packages\\framework-espidf\\components\\esp_system\\ld\r\ngenerator_script: $BUILD_DIR\\esp-idf\\esp_system\\ld\\linker_script_generator.cmake\r\nReading CMake configuration...\r\nDebug: Checking for cmake.exe at: bin\\cmake.exe\r\nDebug: Path exists: False\r\nDebug: Checking for cmake.exe at: C:\\Users\\bigro\\.platformio\\packages\\tool-cmake\\bin\\cmake.exe\r\nDebug: Path exists: True\r\nDebug: Is file: True\r\nDebug: Found cmake.exe at: C:\\Users\\bigro\\.platformio\\packages\\tool-cmake\\bin\\cmake.exe\r\nTypeError: sequence item 1: expected str instance, NoneType found:\r\n File \"C:\\Users\\bigro\\.platformio\\penv\\Lib\\site-packages\\platformio\\builder\\main.py\", line 173:\r\n env.SConscript(\"$BUILD_SCRIPT\")\r\n File \"C:\\Users\\bigro\\.platformio\\packages\\tool-scons\\scons-local-4.8.1\\SCons\\Script\\SConscript.py\", line 620:\r\n return _SConscript(self.fs, *files, **subst_kw)\r\n File \"C:\\Users\\bigro\\.platformio\\packages\\tool-scons\\scons-local-4.8.1\\SCons\\Script\\SConscript.py\", line 280:\r\n exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)\r\n File \"C:\\Users\\bigro\\.platformio\\platforms\\Seeed Studio\\builder\\main.py\", line 23:\r\n env.SConscript(\"board_build/esp/esp_build.py\", exports=\"env\")\r\n File \"C:\\Users\\bigro\\.platformio\\packages\\tool-scons\\scons-local-4.8.1\\SCons\\Script\\SConscript.py\", line 620:\r\n return _SConscript(self.fs, *files, **subst_kw)\r\n File \"C:\\Users\\bigro\\.platformio\\packages\\tool-scons\\scons-local-4.8.1\\SCons\\Script\\SConscript.py\", line 280:\r\n exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)\r\n File \"C:\\Users\\bigro\\.platformio\\platforms\\Seeed Studio\\builder\\board_build\\esp\\esp_build.py\", line 373:\r\n target_elf = env.BuildProgram()\r\n File \"C:\\Users\\bigro\\.platformio\\packages\\tool-scons\\scons-local-4.8.1\\SCons\\Util\\envs.py\", line 252:\r\n return self.method(*nargs, **kwargs)\r\n File \"C:\\Users\\bigro\\.platformio\\penv\\Lib\\site-packages\\platformio\\builder\\tools\\piobuild.py\", line 62:\r\n env.ProcessProgramDeps()\r\n File \"C:\\Users\\bigro\\.platformio\\packages\\tool-scons\\scons-local-4.8.1\\SCons\\Util\\envs.py\", line 252:\r\n return self.method(*nargs, **kwargs)\r\n File \"C:\\Users\\bigro\\.platformio\\penv\\Lib\\site-packages\\platformio\\builder\\tools\\piobuild.py\", line 142:\r\n env.BuildFrameworks(env.get(\"PIOFRAMEWORK\"))\r\n File \"C:\\Users\\bigro\\.platformio\\packages\\tool-scons\\scons-local-4.8.1\\SCons\\Util\\envs.py\", line 252:\r\n return self.method(*nargs, **kwargs)\r\n File \"C:\\Users\\bigro\\.platformio\\penv\\Lib\\site-packages\\platformio\\builder\\tools\\piobuild.py\", line 352:\r\n SConscript(env.GetFrameworkScript(name), exports=\"env\")\r\n File \"C:\\Users\\bigro\\.platformio\\packages\\tool-scons\\scons-local-4.8.1\\SCons\\Script\\SConscript.py\", line 684:\r\n return method(*args, **kw)\r\n File \"C:\\Users\\bigro\\.platformio\\packages\\tool-scons\\scons-local-4.8.1\\SCons\\Script\\SConscript.py\", line 620:\r\n return _SConscript(self.fs, *files, **subst_kw)\r\n File \"C:\\Users\\bigro\\.platformio\\packages\\tool-scons\\scons-local-4.8.1\\SCons\\Script\\SConscript.py\", line 280:\r\n exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)\r\n File \"C:\\Users\\bigro\\.platformio\\platforms\\Seeed Studio\\builder\\frameworks\\espidf.py\", line 1851:\r\n project_codemodel = get_cmake_code_model(\r\n File \"C:\\Users\\bigro\\.platformio\\platforms\\Seeed Studio\\builder\\frameworks\\espidf.py\", line 543:\r\n run_cmake(src_dir, build_dir, extra_args)\r\n File \"C:\\Users\\bigro\\.platformio\\platforms\\Seeed Studio\\builder\\frameworks\\espidf.py\", line 1141:\r\n run_tool(cmd)\r\n File \"C:\\Users\\bigro\\.platformio\\platforms\\Seeed Studio\\builder\\frameworks\\espidf.py\", line 1094:\r\n populate_idf_env_vars(idf_env)\r\n File \"C:\\Users\\bigro\\.platformio\\platforms\\Seeed Studio\\builder\\frameworks\\espidf.py\", line 568:\r\n idf_env[\"PATH\"] = os.pathsep.join(additional_packages + [idf_env[\"PATH\"]])\r\n========================== [FAILED] Took 1.24 seconds =========================="

I have troubleshot this for a very long time and it is because a None type is getting passed to the PATH env variable construction function "populate_idf_env_vars". I added a new function in espidf.py that correctly finds the path for [... cut ...]  which looks right now but populate_idf_env_vars still hangs up on the None type that is being passed into it.

I'm going to keep working on this but has anyone else seen this or is this getting worked on already? I don't want to use the Arduino option because you can't take advantage of all the goodies this chip has to offer since Arduino framework is so basic.

I'm including my modified espidf.py file for you...

[espidf.py.txt](https://github.com/user-attachments/files/19826438/espidf.py.txt)

ackPeng, 2025-04-22:
@robertvandervoort  At present, we have not yet fully ported the configuration of espidf, and the adaptation is all for the Arduino framework; if we want to adapt espidf, we need further efforts.

closed 2025-04-29 by ackPeng

baorepo, 2025-06-12:
@robertvandervoort  espidf + vscode is maintained by Espressif officially and is not part of our content.