Seeed-Studio/platform-seeedboards #5: ESP32C6 +  esp_http_client link error

jose3ww, 2025-05-29:
The compilation of the code below result in link error with "mbedtls" things.

- using Platform IO on MacOS.
- WifiProvisioner related doesn't matter. Just to make it complete (but not tested).
- HTTPClient is not an option because I'd like to use chunk send to save RAM usage.

`
#include <Arduino.h>
#include <list>
#include <WiFi.h>
#include <WiFiProvisioner.h>
#include <esp_http_client.h>

static const char* WEB_SERVER = "http://test.com/upload";
static const char* PASSCODE = "";
static const size_t CHUNK_SIZE = 256;

bool post_chunked(const char* ptr, size_t len)
{
    esp_http_client_config_t config = {
        .url = WEB_SERVER,
    };
    esp_http_client_handle_t client = esp_http_client_init(&config);

    // Transfer-Encoding: chunked
    esp_err_t err = esp_http_client_open(client, -1);
    Serial.println("HTTP: Open connection");
    if (err != ESP_OK) {
        Serial.print("HTTP: Failed to open connection");
        Serial.println(esp_err_to_name(err));
        return false;
    }

    size_t sent_len = 0;
    while(sent_len < len)
    {
        size_t chunk_len = len - sent_len;
        if (chunk_len > CHUNK_SIZE) {
            chunk_len = CHUNK_SIZE;
        }
        Serial.println("HTTP: Sending a chunk");
        int written = esp_http_client_write(client, ptr + sent_len, chunk_len);
        if (written < 0) {
            Serial.println("HTTP: Failed to write a chunk");
            break;
        }
        sent_len += written;
    }

    Serial.println("HTTP: Sending chunks complete");

    // get the HTTP response status code
    int status = esp_http_client_fetch_headers(client);
    Serial.print("HTTP: POST Status = ");
    Serial.println(esp_http_client_get_status_code(client));

    esp_http_client_close(client);
    esp_http_client_cleanup(client);
    return status == 200; // 200 OK
}

void setup() {
  WiFi.mode(WIFI_STA);
  WiFi.begin();

  int retryCount = 0;
  const int maxRetries = 10;
  while (WiFi.status() != WL_CONNECTED && retryCount < maxRetries)
  {
    Serial.print(".");
[... 68 lines cut ...]
/Users/myname/.platformio/packages/toolchain-riscv32-esp@13.2.0+20240530/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: /Users/myname/.platformio/packages/framework-arduinoespressif32/tools/esp32-arduino-libs/esp32c6/lib/libesp-tls.a(esp_tls_mbedtls.c.obj): in function `.L0 ':
(.text.esp_create_mbedtls_handle+0x22): undefined reference to `mbedtls_ssl_init'
/Users/myname/.platformio/packages/toolchain-riscv32-esp@13.2.0+20240530/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: (.text.esp_create_mbedtls_handle+0x34): undefined reference to `mbedtls_ssl_config_init'
/Users/myname/.platformio/packages/toolchain-riscv32-esp@13.2.0+20240530/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: (.text.esp_create_mbedtls_handle+0x82): undefined reference to `mbedtls_ssl_conf_rng'
/Users/myname/.platformio/packages/toolchain-riscv32-esp@13.2.0+20240530/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: (.text.esp_create_mbedtls_handle+0xc0): undefined reference to `mbedtls_ssl_setup'
/Users/myname/.platformio/packages/toolchain-riscv32-esp@13.2.0+20240530/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: (.text.esp_create_mbedtls_handle+0x11e): undefined reference to `mbedtls_ssl_set_bio'
collect2: error: ld returned 1 exit status
*** [.pio/build/seeed_xiao_esp32_c6/firmware.elf] Error 1
`

platform.ini for your reference.
`
[env:seeed_xiao_esp32_c6]
;;; Seeed XIAO ESP32-C6 board configuration
platform = https://github.com/Seeed-Studio/platform-seeedboards.git
; below results in same result 
; platform = Seeed Studio 
board = seeed-xiao-esp32-c6
framework = arduino
monitor_speed = 115200
lib_deps = 
	santerilindfors/WiFiProvisioner@^2.0.0
`

I tried https://github.com/pioarduino/platform-espressif32 too.  It compiled well, but I2C error occured thus not usable.

ackPeng, 2025-06-03:
@jose3ww Hello, I solved this compilation problem by following the steps below: 
1. Download: https://github.com/Jason2866/esp32-arduino-lib-builder and compile according to the readme. After the compilation is completed, there will be an esp32-arduino-libs folder in the out/tools directory. 
2. Delete framework-arduinoespressif32/tools/esp32-arduino-libs and copy the newly compiled folder over.

ackPeng, 2025-06-04:
Already updated

ackPeng, 2025-06-05:
The download link of this package is: https://files.seeedstudio.com/arduino/platformio/framework-arduinoespressif32.zip. The original package was more than 90M, and the updated one should be more than 300M. During the test, I found that the system will cache the old version of the package. If the package you automatically download is still 90M, please manually download and decompress it to the target directory.

ackPeng, 2025-06-05:
Delete the contents under .platformio/.cache/ to avoid cache problems

jose3ww, 2025-06-05:
also it's showing several compilation warnings as follows. Just for your reference.

```
Compiling .pio/build/seeed_xiao_esp32_c6/libf4b/WiFi/STA.cpp.o
/Users/myname/.platformio/packages/framework-arduinoespressif32/libraries/Network/src/NetworkClient.cpp:26: warning: "IN6_IS_ADDR_V4MAPPED" redefined
   26 | #define IN6_IS_ADDR_V4MAPPED(a) ((((__const uint32_t *)(a))[0] == 0) && (((__const uint32_t *)(a))[1] == 0) && (((__const uint32_t *)(a))[2] == htonl(0xffff)))
      | 
In file included from /Users/schino/.platformio/packages/framework-arduinoespressif32/tools/esp32-arduino-libs/esp32c6/include/lwip/lwip/src/include/lwip/sockets.h:53,
                 from /Users/schino/.platformio/packages/framework-arduinoespressif32/tools/esp32-arduino-libs/esp32c6/include/lwip/include/lwip/sockets.h:8,
                 from /Users/schino/.platformio/packages/framework-arduinoespressif32/libraries/Network/src/NetworkClient.cpp:22:
/Users/myname/.platformio/packages/framework-arduinoespressif32/tools/esp32-arduino-libs/esp32c6/include/lwip/lwip/src/include/lwip/inet.h:133: note: this is the location of the previous definition
  133 | #define IN6_IS_ADDR_V4MAPPED(a)     ip6_addr_isipv4mappedipv6((ip6_addr_t*)(a))
      | 

```

```
Compiling .pio/build/seeed_xiao_esp32_c6/FrameworkArduino/esp32-hal-misc.c.o
/Users/myname/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-i2c-slave.c: In function 'i2cSlaveInit':
/Users/myname/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-i2c-slave.c:338:3: warning: implicit declaration of function 'i2c_ll_slave_init'; did you mean 'i2c_ll_slave_rw'? [-Wimplicit-function-declaration]
  338 |   i2c_ll_slave_init(i2c->dev);
      |   ^~~~~~~~~~~~~~~~~
      |   i2c_ll_slave_rw
Compiling .pio/build/seeed_xiao_esp32_c6/FrameworkArduino/esp32-hal-periman.c.o
/Users/myname/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-i2c-slave.c:339:3: warning: implicit declaration of function 'i2c_ll_slave_set_fifo_mode'; did you mean 'i2c_ll_enable_fifo_mode'? [-Wimplicit-function-declaration]
  339 |   i2c_ll_slave_set_fifo_mode(i2c->dev, true);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |   i2c_ll_enable_fifo_mode

```

```
Compiling .pio/build/seeed_xiao_esp32_c6/FrameworkArduino/wiring_pulse.c.o
/Users/myname/.platformio/packages/framework-arduinoespressif32/cores/esp32/freertos_stats.cpp: In function 'void printRunningTasks(Print&)':
/Users/myname/.platformio/packages/framework-arduinoespressif32/cores/esp32/freertos_stats.cpp:72:34: warning: '++' expression of 'volatile'-qualified type is deprecated [-Wvolatile]
   72 |     for (x = 0; x < uxArraySize; x++) {
      |                                  ^
```

ackPeng, 2025-06-05:
@jose3ww  Is there any sample code for me to refer to? This seems to be a toolchain problem. I will compile it with your code.

jose3ww, 2025-06-05:
Here you go.
just add #include <Wire.h>  at the beginning of the file and Wire.begin(); Wire.setClock(400000); at the end of setup();

```
#include <Arduino.h>
#include <Wire.h>
#include <list>
#include <WiFi.h>
#include <WiFiProvisioner.h>
#include <esp_http_client.h>

static const char* WEB_SERVER = "http://test.com/upload";
static const char* PASSCODE = "";
static const size_t CHUNK_SIZE = 256;

bool post_chunked(const char* ptr, size_t len)
{
    esp_http_client_config_t config = {
        .url = WEB_SERVER,
    };
    esp_http_client_handle_t client = esp_http_client_init(&config);

    // Transfer-Encoding: chunked
    esp_err_t err = esp_http_client_open(client, -1);
    Serial.println("HTTP: Open connection");
    if (err != ESP_OK) {
        Serial.print("HTTP: Failed to open connection");
        Serial.println(esp_err_to_name(err));
        return false;
    }

    size_t sent_len = 0;
    while(sent_len < len)
    {
        size_t chunk_len = len - sent_len;
        if (chunk_len > CHUNK_SIZE) {
            chunk_len = CHUNK_SIZE;
        }
        Serial.println("HTTP: Sending a chunk");
        int written = esp_http_client_write(client, ptr + sent_len, chunk_len);
        if (written < 0) {
            Serial.println("HTTP: Failed to write a chunk");
            break;
        }
        sent_len += written;
    }

    Serial.println("HTTP: Sending chunks complete");

    // get the HTTP response status code
    int status = esp_http_client_fetch_headers(client);
    Serial.print("HTTP: POST Status = ");
    Serial.println(esp_http_client_get_status_code(client));

    esp_http_client_close(client);
    esp_http_client_cleanup(client);
    return status == 200; // 200 OK
}

void setup() {
  WiFi.mode(WIFI_STA);
  WiFi.begin();

  int retryCount = 0;
  const int maxRetries = 10;
  while (WiFi.status() != WL_CONNECTED && retryCount < maxRetries)
  {
    Serial.print(".");
    delay(1000);
    retryCount++;
  }

  if (WiFi.status() == WL_CONNECTED)
  {
    Serial.println("\nConnected to WiFi!");
    Serial.println(WiFi.localIP());
    // ここから通常処理
  }
  else
  {
    Serial.println("\nWiFi connect failed. Start provisioning.");
    WiFiProvisioner provisioner;
    provisioner.getConfig().SHOW_INPUT_FIELD = false;
    provisioner.getConfig().SHOW_RESET_FIELD = false;

    // Set the success callback
    provisioner.onSuccess(
        [](const char *ssid, const char *password, const char *input)
        {
          Serial.printf("Provisioning successful! Connected to SSID: %s\n", ssid);
          if (password)
          {
            Serial.printf("Password: %s\n", password);
          }
        });

    // Start provisioning
    Serial.println("Starting WiFi provisioning...");
    provisioner.startProvisioning();

    Serial.println("setting up the sensor connection");
    Wire.begin();          // Start the I2C communication
    Wire.setClock(400000); // Set the I2C clock speed to 400kHz
  }

  uint8_t data[1000];
  for(size_t i = 0; i < 1000; i++)
    data[i] = i;
  post_chunked((char*)&data, 1000);
}

void loop() {
  // put your main code here, to run repeatedly:
}

```

jose3ww, 2025-06-10:
@Yuan-AZKKXL  Thanks! I confirmed my test program and real one both compiles well with the new packages.

closed 2025-07-03 by github-actions[bot]