Seeed-Studio/platform-seeedboards #14: [WARNING] STLink, CMSIS-DAPv2 and PicoProbe probes are not supported because no libusb library was found.

lboue, 2025-09-05:
Hello,

I'm trying to flash my board using my [Raspberry Pi Debug Probe](https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html) but it's not working. 
Is there an additional step with Windows?

```
PS C:\GitHub\Seeed-platform-seeedboards\examples\zephyr-battery> python ..\..\scripts\xiao_nrf54l15_recover_flash.py --hex .pio\build\seeed-xiao-nrf54l15\firmware.hex --mass-erase --probe  COM9
2025-09-05 12:49:42,784 [INFO] Using specified probe: COM9
2025-09-05 12:49:42,784 [INFO] Connecting to target...
2025-09-05 12:49:42,834 [WARNING] STLink, CMSIS-DAPv2 and PicoProbe probes are not supported because no libusb library was found.
Waiting for a debug probe matching unique ID 'COM9' to be connected...
```

lboue, 2025-09-05:
The regular upload is not working anymore so I am trying to find a workaround:

```
 *  Executing task: C:\Users\user\.platformio\penv\Scripts\platformio.exe run --target upload --environment seeed-xiao-nrf54l15 

Processing seeed-xiao-nrf54l15 (platform: https://github.com/Seeed-Studio/platform-seeedboards.git; framework: zephyr; board: seeed-xiao-nrf54l15)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
board id is nrf,will call board_build/nrf/nrf_build.py
CONFIGURATION: https://docs.platformio.org/page/boards/Seeed Studio/seeed-xiao-nrf54l15.html
PLATFORM: Seeed Studio Xiao Series (1.0.0+sha.47379f3) > Seeed Studio XIAO nRF54L15
HARDWARE: NRF54L15 128MHz, 256KB RAM, 1.50MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink)
PACKAGES:
 - framework-zephyr @ 3.40100.0 (4.1.0)
 - tool-bossac-nordicnrf52 @ 1.10901.201022 (1.9.1)
 - tool-cmake @ 3.30.2
 - tool-dtc @ 1.4.7
 - tool-ninja @ 1.9.0 
 - tool-openocd @ 3.1200.0 (12.0)
 - tool-sreccat @ 1.164.0 (1.64)
 - toolchain-gccarmnoneeabi @ 1.80201.181220 (8.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio\build\seeed-xiao-nrf54l15\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   2.3% (used 6084 bytes from 262144 bytes)
Flash: [          ]   1.7% (used 27508 bytes from 1572864 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, custom, jlink
CURRENT: upload_protocol = custom
Uploading .pio\build\seeed-xiao-nrf54l15\firmware.hex
2025-09-05 13:05:42,328 [INFO] Auto-selected probe: 92CC15C9 (Seeed Studio Seeed Studio XIAO nrf54 CMSIS-DAP)
2025-09-05 13:05:42,329 [INFO] Connecting to target...
2025-09-05 13:05:42,429 [INFO] Target type is nrf54l
2025-09-05 13:05:42,483 [INFO] Asserting reset prior to connect
2025-09-05 13:05:42,560 [INFO] DP IDR = 0x6ba02477 (v2 rev6)
2025-09-05 13:05:42,649 [INFO] AHB-AP#0 IDR = 0x84770001 (AHB-AP var0 rev8)
2025-09-05 13:05:42,663 [INFO] AHB-AP#1 IDR = 0x84770001 (AHB-AP var0 rev8)
2025-09-05 13:05:42,673 [INFO] AP#2 IDR = 0x32880000 (AP var0 rev3)
2025-09-05 13:05:42,673 [INFO] Checking CTRL-AP IDR
2025-09-05 13:05:42,697 [WARNING] NRF54L15 APPROTECT enabled: will try to unlock via mass erase
2025-09-05 13:05:43,424 [INFO] AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00fe000 (designer=244 part=01c)
2025-09-05 13:05:43,427 [INFO] [0]<e00ff000:ROM class=1 designer=43b:Arm part=4c9>
2025-09-05 13:05:43,428 [INFO]   AHB-AP#0 Class 0x1 ROM table #1 @ 0xe00ff000 (designer=43b:Arm part=4c9)
2025-09-05 13:05:43,432 [INFO]   [0]<e000e000:SCS M33 class=9 designer=43b:Arm part=d21 devtype=00 archid=2a04 devid=0:0:0>
2025-09-05 13:05:43,435 [INFO]   [1]<e0001000:DWT M33 class=9 designer=43b:Arm part=d21 devtype=00 archid=1a02 devid=0:0:0>
2025-09-05 13:05:43,437 [INFO]   [2]<e0002000:BPU M33 class=9 designer=43b:Arm part=d21 devtype=00 archid=1a03 devid=0:0:0>
2025-09-05 13:05:43,440 [INFO]   [3]<e0000000:ITM M33 class=9 designer=43b:Arm part=d21 devtype=43 archid=1a01 devid=0:0:0>
2025-09-05 13:05:43,448 [INFO]   [5]<e0041000:ETM M33 class=9 designer=43b:Arm part=d21 devtype=13 archid=4a13 devid=0:0:0>
2025-09-05 13:05:43,454 [INFO] [1]<e0040000:TPIU M33 class=9 designer=43b:Arm part=d21 devtype=11 archid=0000 devid=ca1:0:0>
2025-09-05 13:05:43,466 [INFO] CPU core #0: Cortex-M33 r1p0, v8.0-M architecture
2025-09-05 13:05:43,466 [INFO]   Extensions: [DSP, FPU, FPU_V5, MPU, SEC]
2025-09-05 13:05:43,466 [INFO]   FPU present: FPv5-SP-D16-M
2025-09-05 13:05:43,470 [INFO] 4 hardware watchpoints
2025-09-05 13:05:43,472 [INFO] 8 hardware breakpoints, 1 literal comparators
2025-09-05 13:05:43,480 [INFO] Deasserting reset post connect
2025-09-05 13:05:43,488 [ERROR] An error occurred: Memory transfer fault @ 0x00ffc31c-0x00ffc31f
Traceback (most recent call last):
  File "C:\Users\user\.platformio\penv\Lib\site-packages\pyocd\probe\cmsis_dap_probe.py", line 601, in read_ap_result_callback
    value = result()
            ^^^^^^^^
  File "C:\Users\user\.platformio\penv\Lib\site-packages\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 1080, in read_reg_cb
    res = transfer.get_result()
          ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user\.platformio\penv\Lib\site-packages\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 161, in get_result
[... 45 lines cut ...]
  File "C:\Users\user\.platformio\penv\Lib\site-packages\pyocd\core\soc_target.py", line 232, in read_memory
    return self.selected_core_or_raise.read_memory(addr, transfer_size, now)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user\.platformio\penv\Lib\site-packages\pyocd\coresight\cortex_m.py", line 630, in read_memory
    result = self.ap.read_memory(addr, transfer_size, now)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user\.platformio\penv\Lib\site-packages\pyocd\utility\concurrency.py", line 29, in _locking
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... cut ...]
  File "C:\Users\user\.platformio\penv\Lib\site-packages\pyocd\probe\cmsis_dap_probe.py", line 606, in read_ap_result_callback
    raise self._convert_exception(error) from error
pyocd.core.exceptions.TransferFaultError: Memory transfer fault @ 0x00ffc31c-0x00ffc31f
*** [upload] Error 2
==================================================================================================================================== [FAILED] Took 12.97 seconds ====================================================================================================================================

```

StarSphere-1024, 2025-09-10:
Thank you for your feedback. We had a similar problem before, but after using this [script](https://github.com/Seeed-Studio/platform-seeedboards/blob/main/scripts/xiao_nrf54l15_recover_flash.py), it was solved. I'm not sure if your problem is the same as ours. Could you help me test if this other [script](https://gist.github.com/baorepo/9bf8b021589952d515c8792920579b8a) can be uploaded?

You can use the following command:
```
python recover_w_nrf54l15.py --hex /tmp/xiao54-sys_off_rtc_on_merged.hex --mass-erase --probe 12345678
```
Change the probe ID to yours. You can get it by using the command `pyocd list`.

Alternatively, you can try Nordic's official recovery method using J-Link:
Connect the debugging contact at the bottom of the XIAO board with the corresponding pin of the J-Link debugger, and then flash any firmware using J-Flash. This should recover it before flashing.

Toastee0, 2025-09-14:
I posted a new NRF54l15 wipe method here:
https://forum.seeedstudio.com/t/xiao-nrf54l15-an-error-occurred-memory-transfer-fault-0x00ffc31c-0x00ffc31f/293792/28
I used it on one of my own bricked boards, which is now accepting new firmwares.

msfujino, 2025-09-14:
Toastee0,
I will write down the exact steps I took. Is there anything I might have misunderstood?

1. Clean installed Python in C:/Python313
2. Opened Command Prompt as Administrator to verify the path was set and check the version
    C:\Windows\System32>python --version
    Python 3.13.7
3. Next, created a virtual environment in C:/Python313
    and confirmed C:/Python313/.venv existed.
4. Activated the virtual environment.
    C:/Python313> .\.venv\Scripts\activate
    (.env) C:/Python313>
5. Installed libusb.
    (.venv) C:\Python313>pip install libusb
6. Cloned pyOCD
    (.venv) C:\Python313>git clone https://github.com/pyocd/pyOCD.git
7. Changed directory to pyOCD
    (.venv) C:\Python313>cd pyocd
    (.venv) C:\Python313\pyOCD>
8. Run the script
    (.venv) C:\Python313\pyOCD>python -m pyocd erase -u E1799140 -t nrf54l --mass -v
9. Module not found
    ModuleNotFoundError: No module named 'typing_extensions'

lboue, 2025-09-14:
Thanks for the trick but I always get a failure:
```
(.venv) C:\Users\user\Documents\Dev\pyOCD\pyOCD>python -m pyocd erase -u 92CC15C9 -t nrf54l --mass -v
0001403 I Target type is nrf54l [board]
0001466 I DP IDR = 0x6ba02477 (v2 rev6) [dap]
0001494 I AHB-AP#0 IDR = 0x84770001 (AHB-AP var0 rev8) [discovery]
0001514 I AHB-AP#1 IDR = 0x84770001 (AHB-AP var0 rev8) [discovery]
0001524 I AP#2 IDR = 0x32880000 (AP var0 rev3) [discovery]
0001524 I Checking CTRL-AP IDR [target_nRF54L]
0001530 W NRF54L15 APPROTECT enabled: will try to unlock via mass erase [target_nRF54L]
0002254 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00fe000 (designer=244 part=01c) [rom_table]
0002257 I [0]<e00ff000:ROM class=1 designer=43b:Arm part=4c9> [rom_table]
0002257 I   AHB-AP#0 Class 0x1 ROM table #1 @ 0xe00ff000 (designer=43b:Arm part=4c9) [rom_table]
0002261 I   [0]<e000e000:SCS M33 class=9 designer=43b:Arm part=d21 devtype=00 archid=2a04 devid=0:0:0> [rom_table]
0002263 I   [1]<e0001000:DWT M33 class=9 designer=43b:Arm part=d21 devtype=00 archid=1a02 devid=0:0:0> [rom_table]
0002265 I   [2]<e0002000:BPU M33 class=9 designer=43b:Arm part=d21 devtype=00 archid=1a03 devid=0:0:0> [rom_table]
0002268 I   [3]<e0000000:ITM M33 class=9 designer=43b:Arm part=d21 devtype=43 archid=1a01 devid=0:0:0> [rom_table]
0002270 I   [5]<e0041000:ETM M33 class=9 designer=43b:Arm part=d21 devtype=13 archid=4a13 devid=0:0:0> [rom_table]
0002272 I [1]<e0040000:TPIU M33 class=9 designer=43b:Arm part=d21 devtype=11 archid=0000 devid=ca1:0:0> [rom_table]
0002278 I CPU core #0: Cortex-M33 r1p0, v8.0-M architecture [cortex_m]
0002278 I   Extensions: [DSP, FPU, FPU_V5, MPU, SEC] [cortex_m]
0002278 I   FPU present: FPv5-SP-D16-M [cortex_m]
0002280 I 4 hardware watchpoints [dwt]
0002282 I 8 hardware breakpoints, 1 literal comparators [fpb]
0002290 C Memory transfer fault (SWD/JTAG communication failure (FAULT ACK)) @ 0x00ffc31c-0x00ffc31f [__main__]
```

lboue, 2025-09-14:
> 9. Module not found
>    ModuleNotFoundError: No module named 'typing_extensions'

You can use pip install to install dependancies and it will work:
```bash
python -m venv .venv
.\.venv\Scripts\activate
pip install libusb
git clone https://github.com/pyocd/pyOCD.git
cd pyOCD
pip install .
python -m pyocd erase -u CD37EBB4 -t nrf54l --mass -v
python -m pyocd erase -u <Your_ID> -t nrf54l --mass -v
```

To get the ID:
```bash
python -m pyocd list
  #   Probe/Board                                      Unique ID   Target
---------------------------------------------------------------------------
  0   Seeed Studio Seeed Studio XIAO nrf54 CMSIS-DAP   92CC15C9    n/a
```

msfujino, 2025-09-14:
After repeatedly resetting it, reconnecting the USB, and running the script over and over, it suddenly managed to perform a full erase. However, when I run the script again, it sometimes throws an error, sometimes performs a full erase again—it's extremely unstable.

Toastee0, 2025-09-15:
That honestly matches my experience recovering my device! I fussed with that thing and asked around on discords for at least a week before I got this far! But.. I am able to use the one that was unprogrammable before, so congrats on recovering your unit!

StarSphere-1024, 2025-09-16:
A new recovery script is now available here.
It automatically activates the venv, installs the latest PyOCD, and resets your device to factory.

StarSphere-1024, 2025-09-16:
> Regarding my XIAO_nRF54L15, it's highly likely there's a hardware issue. I'm observing signal levels by placing a digital scope probe on the SWDIO and SWCLK pads on the back. The SWDIO pin on the XIAO that cannot upload remains constantly LOW with no changes. The SWDIO pin on the XIAO that can upload is normally HIGH and changes between HIGH and LOW according to the data.

I think nordic's dk board has the same problem when it is locked.But it doesn't matter, now we can easily restore them.

StarSphere-1024, 2025-09-19:
I am glad that you have found the cause of the problem. 
Then close.